pikasTech/PikaPython #28: 关于建立PikaScript的Benchmark的建议

GorgonMeducer, 2022-01-02:
从商业推广和优化的角度考虑，现在急需要为PikaScript建立一个浅显易懂又极具说服力的Benchmark。

推荐如下：
1、选一个主流的平台，比如Cortex-M（推荐以M0和M3为参考，前者是Thumb后者是Thumb2）
2、选一个以穷举为主的算法，推荐计算从0到100之间的素数。这个算法的好处是：
    - 只能穷举
    - 容易检验结果是否正确
    - 包含流控
3、为算法实现对应的 C、Lua、mpy和PikaScript版本
4、统计这些版本所用的周期数，并以C版本为基准来给性能打分：
    假设C所用周期数为 A，PikaScript 所用的周期数为B，则性能评分为：  （A/B * 100 %）
    表示对应的脚本算法性能是C语言的百分比。
5、建议测试 -O0、-Oz、-Os和-Ofast 下的跑分（以GCC为蓝本）
6、将Lua、mpy和PikaScript的跑分、所需的资源等信息列成表格，供大家参考。

pikasTech, 2022-01-03:
先在CI搞一个和C的对比跑分

GorgonMeducer, 2022-01-03:
是的，这是一个不错的开始。

pikasTech, 2022-01-03:
> 是的，这是一个不错的开始。

一股翻译腔

pikasTech, 2022-01-03:
benchmark on Linux is ok
the figure of benchmark result would be printed to the website when CI is running.
https://pikastech.github.io/pikascript/dev/bench/
[image]

the code of the benchmark(a big for-loop):
https://github.com/pikasTech/pikascript/blob/master/port/linux/benchmark/py.cpp

GorgonMeducer, 2022-01-03:
测试用的 py真够敷衍的。话说横轴是啥？看不懂呢。

另外，我们要的 跟C语言相比的百分比呢？

pikasTech, 2022-01-03:
刚跑通流程，后面的慢慢加（捂脸
横轴是commit

pikasTech, 2022-01-04:
prime number test is ok

[image]

cleck here to see the latest benchmark

[image]

pikasTech, 2022-01-04:
The 100 prime numbers within the C language were added, and the benchmark was calibrated

closed 2022-01-04 by GorgonMeducer