pikasTech/PikaPython #90: Linux doesn't build

maxgerhardt, 2022-04-07:
When I follow https://github.com/pikasTech/pikascript#6test-and-develop-core, I get all sorts of errors which I have to correct.

```
max@max-VirtualBox:~/pika/pikascript/port/linux$ sh make.sh # Compile the project.
api-make-linux.sh: 2: ./rust-msc-latest-linux: not found
api-make-win10.sh: 2: ./rust-msc-latest-win10.exe: Permission denied
make.sh: 2: cd: can't cd to build
make: *** No targets specified and no makefile found.  Stop.
```

And then
```
max@max-VirtualBox:~/pika/pikascript/port/linux$ cp ../../tools/pikaCompiler/rust-msc-latest-linux package/pikascript/
```
after that
```
max@max-VirtualBox:~/pika/pikascript/port/linux$ sh make.sh
analyzing file: main.py
analyzing file: PikaStdLib.py
analyzing file: PikaStdData.py
analyzing file: GTestTask.py
analyzing file: PikaStdTask.py
analyzing file: PikaStdData.py
analyzing file: PikaMath.py
analyzing file: PikaStdDevice.py
analyzing file: PikaDebug.py
analyzing file: PikaStdData.py
api-make-win10.sh: 2: ./rust-msc-latest-win10.exe: Permission denied
make.sh: 2: cd: can't cd to build
make: *** No targets specified and no makefile found.  Stop. 
```

But when I do

```
cp ../../tools/pikaCompiler/rust-msc-latest-win10.exe package/pikascript/.
max@max-VirtualBox:~/pika/pikascript/port/linux$ chmod +x package/pikascript/rust-msc-latest-*
max@max-VirtualBox:~/pika/pikascript/port/linux$ sh make.sh
analyzing file: main.py
analyzing file: PikaStdLib.py
analyzing file: PikaStdData.py
analyzing file: GTestTask.py
analyzing file: PikaStdTask.py
analyzing file: PikaStdData.py
analyzing file: PikaMath.py
analyzing file: PikaStdDevice.py
analyzing file: PikaDebug.py
analyzing file: PikaStdData.py
api-make-win10.sh: 2: ./rust-msc-latest-win10.exe: Exec format error
make.sh: 2: cd: can't cd to build
make: *** No targets specified and no makefile found.  Stop.
```

pikasTech, 2022-04-08:
`
sh install_dependency.sh # install dependency 
sh pull-core.sh
sh init.sh # Only nedded in the first time.
sh make.sh # Compile the project.
`
you can try: 
use sh pull-core.sh and use sh init.sh again.

And the readme maybe out of date.
The most usable build steps is in the CI scripts

https://github.com/pikasTech/pikascript/blob/master/.github/workflows/CI.yml

maxgerhardt, 2022-04-08:
After applying #91 and #92 and rerunning instructions per updated readme, the `init.sh` script still fails for the Win10 case (?)

```
api-make-win10.sh: 2: ./rust-msc-latest-win10.exe: Permission denied
```
and before that it throws

```
(pikascript) packages installed:
    'requestment.txt' no found.
```

But then `make.sh` seems to go through except again for the Windows 10 part

```
max@max-VirtualBox:~/pika/pikascript/port/linux$ sh make.sh 
(pikascript) packages installed:
    'requestment.txt' no found.

(pikascript) pika compiler:
    compiling main.py...
    compiling PikaStdLib.py...
    compiling PikaStdData.py...
    compiling GTestTask.py...
    compiling PikaStdTask.py...
    compiling PikaStdData.py...
    compiling PikaMath.py...
    compiling PikaStdDevice.py...
    compiling PikaDebug.py...
    compiling PikaStdData.py...
    compiling PikaStdTask.py...
    compiling PikaStdData.py...
    compiling PikaStdData.py...
    compiling PikaDebug.py...

api-make-win10.sh: 2: ./rust-msc-latest-win10.exe: Permission denied
[51/51] Linking CXX executable test/pikascript_test
```

Unit tests and benchmarks then seem to run normally though.

pikasTech, 2022-04-08:
Yes，some error about win10 can be ignored...

And the these error info would be clear soon

closed 2022-04-08 by maxgerhardt