pikasTech/PikaPython #93: Compiling STM32F103C8 booter from Makefile

maxgerhardt, 2022-04-08:
The repo says

https://github.com/pikasTech/pikascript/blob/5d5d4974d56f1b6482d7d521d054768f8cd06279/bsp/stm32f103c8/Makefile#L61-L65

but the folder doesn't have a `Core/src` folder, only `Boot`. How can this compile?

pikasTech, 2022-04-08:
The keil-mdk project is mainly maintained.
I will fix the makefile of STM32F103C8 bsp. 
But for other bsps, the makefile maybe be remove.

closed 2022-04-08 by pikasTech

pikasTech, 2022-04-09:
@maxgerhardt 

Now the community edition of MDK-ARM is released. If there is no commercial requirement, it is free like GCC!

https://www2.keil.com/mdk5/editions/community

And the flash size can be reduced by 50% with MDK-ARM(AC6)!

There is also an MDK-ARM template for Raspberry Pi Pico, it's amazing.

https://github.com/GorgonMeducer/Pico_Template

maxgerhardt, 2022-04-09:
I had an idea to add a `platformio.ini` and a small builder script so that all embedded firmwares can be built + flashed with PlatformIO, which would then use GCC. Still working on that / figuring out the best way. Just so that people have an option, especially since Keil is a Windows-only software (https://developer.arm.com/documentation/ka002387/latest) and Linux and Mac people would have to use a virtual machine + Windows to run this software :( 

Of course it would then be interesting to see what code sizes and ram usage GCC can do vs the Keil compiler.

pikasTech, 2022-04-09:
> 

Haha, Linux is amazing too.