pikasTech/PikaPython #94: Python shell is really broken

maxgerhardt, 2022-04-08:
Based on the code from pikascript.com and all `==latest` versions for the STM32F103C8 and a slight adaption to call into `pikaScriptShell(pikaMain)` and giving the firmware the needed `char __platform_getchar(void)` function, I successfully get a Python shell on my Bluepill board.

```
>miniterm.py -e --eol CRLF COM6 115200
======[pikascript packages installed]======
PikaStdDevice==latest
PikaStdLib==latest
STM32F1==latest
pikascript-core==latest
===========================================
memory used now: 140 B
memory used max: 848 B

>>>
>>>
```

While some functions work

```
>>> 2**8
256.000000
>>> "A" + "B"
AB
>>> print("Hello" + " world")
Hello world
>>> x = 3.0
>>> y = 5.0
>>> x +y
8.000000
```
Others are really broken

```
>>> 1+1
0
>>> 20 + 30
0
>>> 2 * 3
0
>>> int("3")
[error] convert to int type faild.
   STR 3                (#1)
 -> RUN int             (#3)
>>> print("A" if True else "False")
AifTrueelseFalse
>>> len("ABC")
NameError: name 'len' is not defined
   STR ABC              (#1)
 -> RUN len             (#5)
```
Is there something wrong with how I build the shell or did I miss something else?

The project is at https://github.com/maxgerhardt/pikascript-pio-bluepill/ and is using PlatformIO, very easy to use + upload + debug.

maxgerhardt, 2022-04-08:
Examples like https://github.com/pikasTech/pikascript/blob/master/examples/GPIO/main.py only partially work

```py
>>> import PikaStdLib
>>> import machine
>>> mem = PikaStdLib.MemChecker()
>>> io1 = machine.GPIO()
Error: method 'machine.GPIO' no found.
 -> RUN machine.GPIO            (#1)
>>> time = machine.Time()
Error: method 'machine.Time' no found.
 -> RUN machine.Time            (#1)
>>> print('hello pikascript')
hello pikascript
>>> print('mem.max :')
mem.max :
>>> mem.max()
1.83 kB
>>> print('mem.now :')
mem.now :
>>> mem.now()
0.96 kB
```

pikasTech, 2022-04-08:
> Based on the code from pikascript.com and all `==latest` versions for the STM32F103C8 and a slight adaption to call into `pikaScriptShell(pikaMain)` and giving the firmware the needed `char [...]

- Build-in functions like 'len()' and 'int()' now is in building. And after some fix, the len() and int() now can support str.
- But the '1+1', '20+30', '2*3" run fine in linux. Maybe some wrong with it when it run in Bluepill. It need time to figure it out.
```
>>> 1+1
2
>>> 20+30
50
>>> 2*3
6
>>> int("3")
3
>>> len("ABC")
3
```
- The advanced syntax now is no supported. For the 'if' 'else', only the basic usage is supported.
```
if a > 0:
    print(a)
else:
    print(b)
```

pikasTech, 2022-04-08:
> Examples like https://github.com/pikasTech/pikascript/blob/master/examples/GPIO/main.py only partially work
> 
> ```python
> >>> import PikaStdLib
> >>> import machine
> >>> mem = PikaStdLib.MemChecker()
> >>> io1 = machine.GPIO()
> Error: method 'machine.GPIO' no found.
>  -> RUN machine.GPIO            (#1)
> >>> time = machine.Time()
> Error: method 'machine.Time' no found.
>  -> RUN machine.Time            (#1)
> >>> print('hello pikascript')
> hello pikascript
> >>> print('mem.max :')
> mem.max :
> >>> mem.max()
> 1.83 kB
> >>> print('mem.now :')
> mem.now :
> >>> mem.now()
> 0.96 kB
> ```

Becaue there is no machine.py in the 'pikascript' folder.
A 'pikascript' folder is not only containe the C source code. 

It also need python files and tools. And the files in 'pikascript-api' are generated by preCompiler based on the python files.

A typical 'pikascript' folder is here:

https://github.com/pikasTech/pikascript/tree/master/bsp/stm32g030c8/pikascript

The [requestment.txt](https://github.com/pikasTech/pikascript/blob/master/bsp/stm32g030c8/pikascript/requestment.txt) discribed the packages to be installed. 

The pikaPackage.exe can install the packages based on the requestment.txt

The [main.py](https://github.com/pikasTech/pikascript/blob/master/bsp/stm32f103c8/pikascript/main.py) is the entry of pikascript python script.

And [machine.py](https://github.com/pikasTech/pikascript/blob/master/bsp/stm32g030c8/pikascript/machine.py) is only the skin of 'STM32G0'. You can make a machine.py youself and inherit from STM32F1.py. Or just ` import STM32F1 ` instead.

maxgerhardt, 2022-04-08:
I found the bug: You are printing a `int64_t` with `%d`, although that classifies as a long long integer! So in ARM, two 32-bit words (2x32-bit part of the 64-bit integer) are pushed onto the stack, but only one is popped back due to the `%d` = int = 32-bit specifier, and so only the higher 32-bits (which are 0 in the 1+1 example) are printed, leading to overall 0.

Since newlib-nano (`--specs=nano.specs`) doesn't support `%lld`, I casted the return value of `arg_getInt()` to `(int)` and then printed it, and then...

```
>>> 1 + 1
2
>>> 20 + 30
50
```

maxgerhardt, 2022-04-08:
How could this work on the STM32G0 board you're testing with? Anyways I suggest changing

https://github.com/pikasTech/pikascript/blob/757f00893a9475a0073e512828cd21ae31a459a7/src/PikaVM.c#L1253-L1255

to use platform-independent and correct macros for printing that `int64_t` (like PRIi64 or something) and double checking all embedded boards are even capable of that printf option, especially if they're compiled with newlib-nano.. Or downcasting the result / polyfilling the missing functionality.

maxgerhardt, 2022-04-08:
Hm, doing 

```
#include <inttypes.h>
...
__platform_printf("%" PRIi64 "\r\n", arg_getInt(arg));
```
Does not work with arm-none-eabi-gcc + `--specs=nano.specs`, it expands to `lli` but does not react correctly to it, jus prints `li` on the console. I fear this might need the full C library and not the newlib-nano version of it, but this will make the firmware size explode.. Or maybe it's better to introduce a switch to limit printing to 64 bit ints. Or throw in a better general printf function like https://github.com/ARMmbed/mbed-os/tree/master/platform/source/minimal-printf, which *does* support 64-bit integers..

pikasTech, 2022-04-08:
> How could this work on the STM32G0 board you're testing with? Anyways I suggest changing [...]

It's a new bug from v1.6.0. Thank you for figuring out it. I did not find it when I test STM32G0. Transfer arg_getInt(arg) to int32 may be a good idea.

pikasTech, 2022-04-08:
> Hm, doing
>
> ```
> #include <inttypes.h>
> ...
> __platform_printf("%" PRIi64 "\r\n", arg_getInt(arg));
> ```
>
> Does not work with arm-none-eabi-gcc + `--specs=nano.specs`, it expands to `lli` but does not [...]

64bit is too complex ...

maxgerhardt, 2022-04-08:
In any case in the future it would need some big-numbers library because Python is supposed to do arbitrary-length integer arithmetic isn't it? 😅 

Another option would be to pull in the convert `int64_t` to string function from minimal-printf from somewhere else and then use `%s`. That would support all platforms well.

pikasTech, 2022-04-08:
> In any case in the future it would need some big-numbers library because Python is supposed to do arbitrary-length integer arithmetic isn't it? sweat_smile
> 
> Another option would be to pull in the convert `int64_t` to string function from minimal-printf from somewhere else and then use `%s`. That would support all platforms well.

I used the int32 now.

https://github.com/pikasTech/pikascript/blob/master/src/PikaVM.c#L1254

At the same time, It's true that a cross-platform printf is necessary. This should be a TODO issue. #95

maxgerhardt, 2022-04-08:
For now with b2e77ddcf09a169bcbfa7187aa1f406ac466ac21 and 757f00893a9475a0073e512828cd21ae31a459a7 I see this issue as closed, I also confirmed it working in my project.

closed 2022-04-08 by maxgerhardt