pikasTech/PikaPython #96: Loop problems

maxgerhardt, 2022-04-08:
With my [referenced project](https://github.com/maxgerhardt/pikascript-pio-bluepill)

```python
>>> for i in range(0, 10):
... print(i)
[error] print: can not print val
   REF i                (#56)
 -> RUN print           (#63)
```

And while loops hang up completely

```python
>>> i = 0
>>> i
0
>>> while i != 10:
... i +=1

<hangs>
```

maxgerhardt, 2022-04-08:
When okay that might have been due to missing spaces in the second line, but with them I just get syntax error.

```py
>>> for i in range(0, 10):
...    print(i)
... 
[error]: Syntax error.
```

pikasTech, 2022-04-09:
only support 4 space

pikasTech, 2022-04-09:
> With my [referenced project](https://github.com/maxgerhardt/pikascript-pio-bluepill)
>
> ```python
> ... print(i)
> [error] print: can not print val
>    REF i                (#56)
>  -> RUN print [...]

``` python
Python 3.10.4 (main, Apr  2 2022, 09:04:19) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> while a >0:
... 
  File "<stdin>", line 2
    
    ^
IndentationError: expected an indented block after 'while' statement on line 1
```

No space is also wrong in CPython. The robustness of PikaScript is not good enough for wrong input, so checking it in CPython first may be helpful.

maxgerhardt, 2022-04-09:
Indeed you're right, the 4 spaces is really important.

```py
>>> for i in range(0,10):
...     print(i)
...
0
1
2
3
4
5
6
7
8
9
>>> i = 0
>>> while i != 10:
...     i += 1
...
>>> i
10
```
But maybe throwing an `IndentationError` would be better.

Also things that would be syntax errors are not marked as one but just execute

```py
>>> while i != 10
>>> 
```
vs
```py
>>> while i != 10
  File "<stdin>", line 1
    while i != 10
                ^
SyntaxError: invalid syntax
```

But anyways this error was on my side.

closed 2022-04-09 by maxgerhardt

pikasTech, 2022-04-09:
> Indeed you're right, the 4 spaces is really important.
> 
> ```python
> >>> for i in range(0,10):
> ...     print(i)
> ...
> 0
> 1
> 2
> 3
> 4
> 5
> 6
> 7
> 8
> 9
> >>> i = 0
> >>> while i != 10:
> ...     i += 1
> ...
> >>> i
> 10
> ```
> 
> But maybe throwing an `IndentationError` would be better.
> 
> Also things that would be syntax errors are not marked as one but just execute
> 
> ```python
> >>> while i != 10
> >>> 
> ```
> 
> vs
> 
> ```python
> >>> while i != 10
>   File "<stdin>", line 1
>     while i != 10
>                 ^
> SyntaxError: invalid syntax
> ```
> 
> But anyways this error was on my side.

It's true that the error checker is need enhancement.