pikasTech/PikaPython #97: List Problems

maxgerhardt, 2022-04-08:
Same setup as in #96 
```py
>>> myList = [1,2,3]
NameError: name '[1,2,__get__' is not defined
   NUM 3                (#1)
 -> RUN [1,2,__get__            (#3)
```
Really weird error messages too.

maxgerhardt, 2022-04-08:
Hm I get the feeling this is because I don't import the `PikaStdData.py` which declares the `List` class? But if the shell is running, how do I import this file? I can't even define a function due to syntax errors.

```py
>>> class GPIO(TinyObj):
...    def __init__(self):
...       pass
...    def append(self, arg: any):
...       pass
... 
[error]: Syntax error.
```

pikasTech, 2022-04-09:
import in runtime is not suportted

just import in main.py then you can use it in shell

pikasTech, 2022-04-09:
class and function define is only supported in main.py

pikasTech, 2022-04-09:
> 

The List() is only supplied by the PikaStdData.List() class and the document are here.

https://pikadoc.readthedocs.io/zh/latest/PikaStdData%20%E6%95%B0%E6%8D%AE%E7%BB%93%E6%9E%84.html#class-list

maxgerhardt, 2022-04-09:
Right so after https://github.com/maxgerhardt/pikascript-pio-bluepill/commit/82c7e4bc3c3fe40f42d306a56dc6d57d8294ea45,   I can do 

```py
>>> x = PikaStdData.List()
>>> x.len()
0
>>> x.append(1)
>>> x.append(2)
>>> x
>>>
>>> for elem in x:
...     print(elem)
...
1
2
```

Which works well here, but importing it into the global namespace seems to not work :/ 

```py
>>> from PikaStdData import List
>>> x = List()
NameError: name 'List' is not defined
 -> RUN List            (#1)
>>> from PikaStdData import *
>>> x = List()
NameError: name 'List' is not defined
 -> RUN List            (#1)
```

And shouldn't `list()`, `dict()`, `bytearray()` and `string()` be a built-in type and always there?

pikasTech, 2022-04-09:
> Right so after [maxgerhardt/pikascript-pio-bluepill@82c7e4b](https://github.com/maxgerhardt/pikascript-pio-bluepill/commit/82c7e4bc3c3fe40f42d306a56dc6d57d8294ea45), I can do
> 
> ```python
> >>> x = PikaStdData.List()
> >>> x.len()
> 0
> >>> x.append(1)
> >>> x.append(2)
> >>> x
> >>>
> >>> for elem in x:
> ...     print(elem)
> ...
> 1
> 2
> ```
> 
> Which works well here (thanks!), but importing it into the global namespace seems to not work :/
> 
> ```python
> >>> from PikaStdData import List
> >>> x = List()
> NameError: name 'List' is not defined
>  -> RUN List            (#1)
> >>> from PikaStdData import *
> >>> x = List()
> NameError: name 'List' is not defined
>  -> RUN List            (#1)
> ```
> 
> And shouldn't `list()`, `dict()`, `bytearray()` and `string()` be a built-in type and always there? Even after import, native lists still throw the weird error
> 
> ```python
> >>> x = [1,2,3]
> NameError: name '[1,2,__get__' is not defined
>    NUM 3                (#1)
>  -> RUN [1,2,__get__            (#3)
> ```

> Right so after [maxgerhardt/pikascript-pio-bluepill@82c7e4b](https://github.com/maxgerhardt/pikascript-pio-bluepill/commit/82c7e4bc3c3fe40f42d306a56dc6d57d8294ea45), I can do
> 
> ```python
> >>> x = PikaStdData.List()
> >>> x.len()
> 0
> >>> x.append(1)
> >>> x.append(2)
> >>> x
> >>>
> >>> for elem in x:
> ...     print(elem)
> ...
> 1
> 2
> ```
> 
> Which works well here (thanks!), but importing it into the global namespace seems to not work :/
> 
> ```python
> >>> from PikaStdData import List
> >>> x = List()
> NameError: name 'List' is not defined
>  -> RUN List            (#1)
> >>> from PikaStdData import *
> >>> x = List()
> NameError: name 'List' is not defined
>  -> RUN List            (#1)
> ```
> 
> And shouldn't `list()`, `dict()`, `bytearray()` and `string()` be a built-in type and always there? Even after import, native lists still throw the weird error
> 
> ```python
> >>> x = [1,2,3]
> NameError: name '[1,2,__get__' is not defined
>    NUM 3                (#1)
>  -> RUN [1,2,__get__            (#3)
> ```

This is for trimmability reasons, in some applications list() and dict() are not used, then not introducing them can help reduce the firmware size.

And `from * import *` is now not supported. Maybe usable in future.

https://pikadoc.readthedocs.io/en/latest/PikaScript%20%E6%A8%A1%E5%9D%97%E6%A6%82%E8%BF%B0.html#import-and-call-the-module

maxgerhardt, 2022-04-09:
Maybe the core should have a `#define` / macro option to enable built in lists, dictionaries, etc. as a selectable featureset. I think it's with regards to Python programming that things like `x = [1,2,3]` work built-in, also with e.g. `len(x)` then, which with the `List` class hes to be done with `x.len()`.

pikasTech, 2022-04-09:
> Maybe the core should have a `#define` / macro option to enable built in lists, dictionaries, etc. as a selectable featureset. I think it's with regards to Python programming that things like `x = [...]

Okay, the build-in mode for `list()` `dict()` shoud be a TODO issue.

pikasTech, 2022-04-14:
Now https://github.com/pikasTech/pikascript/commit/72a28aae77497186248e0e7067523188b31dd9f1 you can use the `PIKA_BUILTIN_LIST_ENBALE` and `PIKA_BUILTIN_DICT_ENBALE` to enable built-in `list` and `dict`.
And now you can do that with `list`:
```
>>> x = [1,2,3]
>>> for item in x:
...     item
... 
1
2
3
>>> 
```
For `dict`, you can only use `x = dict()` to create a `dict`.

About how to config pikascript:

https://pikadoc.readthedocs.io/en/latest/%E4%BC%98%E5%8C%96%E5%86%85%E5%AD%98%E5%8D%A0%E7%94%A8%E3%80%81%E9%85%8D%E7%BD%AE%20libc.html

pikasTech, 2022-04-14:
@maxgerhardt there is a typo about the macro and now fixed. https://github.com/pikasTech/pikascript/commit/f6f38bdb7378fa27af325bfb05289e6a7aaf9b65

closed 2022-04-15 by pikasTech