pikasTech/PikaPython #98: Function Definition Problems

maxgerhardt, 2022-04-08:
In the script console, I can't seem to be able to define a function that will be executed later due to a syntax error (?)

```py
[error]: Syntax error.
>>> def a():
...    print("A")
...    print("B")
... 
[error]: Syntax error.
>>>
>>> a()
>>> 
```

This code works in a Python shell as regular.

```py
>>> def a():
...    print("A")
...    print("B")
...
>>> a()
A
B
```

maxgerhardt, 2022-04-08:
Aha, with 4 spaces instead of 3 (which python did not need), I can get a function definition but it still won't execute correctly.

```py
>>> def a():
...     print("A")
... 
>>> a()
```

pikasTech, 2022-04-09:
function define is only supported in main.py

maxgerhardt, 2022-04-09:
Should we leave this issue open as "Support function definition in pikaScriptShell?

pikasTech, 2022-04-09:
> Should we leave this issue open as "Support function definition in pikaScriptShell?

Okay, it can be a TODO issue for the next release version. #88

closed 2022-04-11 by pikasTech

maxgerhardt, 2022-04-11:
Confirmed solved after update.