pikasTech/PikaPython #100: No heap space! Please reset the device.

1qyl1, 2022-04-11:
[image]
task.call_period_ms(test,500)
task.run_forever()
[error]: No heap space! Please reset the device.

大佬，，这个怎么解决啊，，。

main.py
def test():
    print(1)

pikasTech, 2022-04-11:
```
def test():
print(1)
```
这个写法是语法错误的，def 后面不能不带缩进，pika 还不能很稳定地处理错误输入。请先在 cpython 上验证基础语法的正确性。

其次，也有可能就是 heap 的大小不过了，请检查 heap 的大小，以及使用 `mem.max()` 检查最大的内存占用。

考虑到内存碎片的影响，如果 `mem.max` 已经达到 heap 区的 70% 左右，是有可能 heap 太小的。

1qyl1, 2022-04-11:
[image]
首先 这个 test函数是没问题的，在shell里面可以直接使用，cpython也没问题。
[image]
然后由于 不能打印浮点 我把结果强转为int打印 
>>> mem.max()
3 kB
>>> mem.max()
3 kB
>>> mem.now()
0 kB
[image]

然后 na那个 heap 的问题，rtt怎么管理软件包的heap呢  如何修改啊，，

pikasTech, 2022-04-11:
> > ```
> > def test():
> > print(1)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 这个写法是语法错误的，def 后面不能不带缩进，pika 还不能很稳定地处理错误输入。请先在 cpython 上验证基础语法的正确性。
> > 其次，也有可能就是 heap 的大小不够了，请检查 heap 的大小，以及使用 `mem.max()` 检查最大的内存占用。
> > 考虑到内存碎片的影响，如果 `mem.max()` 已经达到 heap 区的 70% 左右，则有可能的确是 heap 太小导致的。
> 
> 我的芯片ram350+ 就运行了 rtt系统和pika

请给出完整信息：

包括：
- 安装的 pikascript package 的版本
- 完整的 shell log，尤其包括创建 task() 对象的

1qyl1, 2022-04-11:
> > > ```
> > > def test():
> > > print(1)
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 这个写法是语法错误的，def 后面不能不带缩进，pika 还不能很稳定地处理错误输入。请先在 cpython 上验证基础语法的正确性。
> > > 其次，也有可能就是 heap 的大小不够了，请检查 heap 的大小，以及使用 `mem.max()` 检查最大的内存占用。
> > > 考虑到内存碎片的影响，如果 `mem.max()` 已经达到 heap 区的 70% 左右，则有可能的确是 heap 太小导致的。
> > 
> > 
> > 我的芯片ram350+ 就运行了 rtt系统和pika
> 
> 请给出完整信息：
> 
> 包括：
> 
> * 安装的 pikascript packages 的版本，如果不是太老的版本，那么启动时会全部打印出来
> * 完整的 shell log，尤其包括创建 task() 对象的

[image]
[image]
[image]
[image]

pikasTech, 2022-04-11:
> [image] 首先 这个 test函数是没问题的，在shell里面可以直接使用，cpython也没问题。 [image] 然后由于 不能打印浮点 我把结果强转为int打印
> 
> > > > mem.max()
> > > > 3 kB
> > > > mem.max()
> > > > 3 kB
> > > > mem.now()
> > > > 0 kB
> > > > [image]
> 
> 然后 na那个 heap 的问题，rtt怎么管理软件包的heap呢 如何修改啊，，

关于无法打印浮点的问题，可以通过 `getMax()` api 来得到整形的内存值（字节单位）

1qyl1, 2022-04-11:
> 关于 rtt 的 heap 设置问题，请参考 rtt 的文档，这个不在我负责的范围内。
不过我认为不应该是heap大小的问题    我会持续关注的。。感谢大佬的帮助   。

pikasTech, 2022-04-11:
> > 关于 rtt 的 heap 设置问题，请参考 rtt 的文档，这个不在我负责的范围内。
> > 不过我认为不应该是heap大小的问题    我会持续关注的。。感谢大佬的帮助   。

我在 pikaRTThread v1.2.0 中加入了 Task() 类，然后在qemu进行了测试，可以正常工作，你可以更新以下 pikaRTThread 然后试试。

``` python
#main.py
import pikaRTThread
import PikaStdLib

mem = PikaStdLib.MemChecker()

def todo():
    mem.max()

```

```
D:\RT-ThreadStudio\workspace\A9>D:/RT-ThreadStudio/repo/Extract/Debugger_Support_Packages/RealThread/QEMU/4.2.0.4/qemu-system-arm.exe -M vexpress-a9 --kernel Debug/rtthread.bin -nographic -show-cursor
em-arm.exe -M vexpress-a9 --kernel Debug/rtthread.bin -nographic -show-cursor

 \ | /
- RT -     Thread Operating System
 / | \     4.0.2 build Apr 11 2022
 2006 - 2019 Copyright by rt-thread team
lwIP-2.0.2 initialized!
[I/sal.skt] Socket Abstraction Layer initialize success.
hello rt-thread
msh />pika
------------------------------------------------------------------
|                                                                |
|     ____   _   __            _____              _          __  |
|    / __ \ (_) / /__ ____ _  / ___/ _____ _____ (_) ____   / /_ |
|   / /_/ // / / //_// __ `/  \__ \ / ___// ___// / / __ \ / __/ |
|  / ____// / / ,<  / /_/ /  ___/ // /__ / /   / / / /_/ // /_   |
| /_/    /_/ /_/|_| \__,_/  /____/ \___//_/   /_/ / .___/ \__/   |
|                                                /_/             |
|          PikaScript - An Ultra Lightweight Python Engine       |
|                                                                |
|           [ https://github.com/pikastech/pikascript ]          |
|           [  https://gitee.com/lyon1998/pikascript  ]          |
|                                                                |
------------------------------------------------------------------
======[pikascript packages installed]======
PikaStdDevice==v1.6.0
PikaStdLib==v1.6.2
pikaRTDevice==v1.0.1
pikaRTThread==v1.2.0
pikascript-core==v1.6.2
===========================================
>>> task = pikaRTThread.Task()
>>> task.call_period_ms(todo,500)
>>> task.run_forever()
3.13 kB
3.24 kB
3.24 kB

```

1qyl1, 2022-04-14:
> > > 关于 rtt 的 heap 设置问题，请参考 rtt 的文档，这个不在我负责的范围内。
> > > 不过我认为不应该是heap大小的问题    我会持续关注的。。感谢大佬的帮助   。
> 
> 我在 pikaRTThread v1.2.0 中加入了 Task() 类，然后在qemu进行了测试，可以正常工作，你可以更新一下 pikaRTThread 然后试试。
> 
> ```python
> #main.py
> import pikaRTThread
> import PikaStdLib
> 
> mem = PikaStdLib.MemChecker()
> 
> def todo():
>     mem.max()
> ```
> 
> ```
> D:\RT-ThreadStudio\workspace\A9>D:/RT-ThreadStudio/repo/Extract/Debugger_Support_Packages/RealThread/QEMU/4.2.0.4/qemu-system-arm.exe -M vexpress-a9 --kernel Debug/rtthread.bin -nographic -show-cursor
> em-arm.exe -M vexpress-a9 --kernel Debug/rtthread.bin -nographic -show-cursor
> 
>  \ | /
> - RT -     Thread Operating System
>  / | \     4.0.2 build Apr 11 2022
>  2006 - 2019 Copyright by rt-thread team
> lwIP-2.0.2 initialized!
> [I/sal.skt] Socket Abstraction Layer initialize success.
> hello rt-thread
> msh />pika
> ------------------------------------------------------------------
> |                                                                |
> |     ____   _   __            _____              _          __  |
> |    / __ \ (_) / /__ ____ _  / ___/ _____ _____ (_) ____   / /_ |
> |   / /_/ // / / //_// __ `/  \__ \ / ___// ___// / / __ \ / __/ |
> |  / ____// / / ,<  / /_/ /  ___/ // /__ / /   / / / /_/ // /_   |
> | /_/    /_/ /_/|_| \__,_/  /____/ \___//_/   /_/ / .___/ \__/   |
> |                                                /_/             |
> |          PikaScript - An Ultra Lightweight Python Engine       |
> |                                                                |
> |           [ https://github.com/pikastech/pikascript ]          |
> |           [  https://gitee.com/lyon1998/pikascript  ]          |
> |                                                                |
> ------------------------------------------------------------------
> ======[pikascript packages installed]======
> PikaStdDevice==v1.6.0
> PikaStdLib==v1.6.2
> pikaRTDevice==v1.0.1
> pikaRTThread==v1.2.0
> pikascript-core==v1.6.2
> ===========================================
> >>> task = pikaRTThread.Task()
> >>> task.call_period_ms(todo,500)
> >>> task.run_forever()
> 3.13 kB
> 3.24 kB
> 3.24 kB
> ```

大佬 我发现个问题  

[image]

这是d当前内存的占用值   ，，一运行 task.run_forever()函数后  heap就溢出，，，，可能是什么原因造成的呢。。

pikasTech, 2022-04-14:
> 大佬 我发现个问题
> 
> [image]
> 
> 这是d当前内存的占用值 ，，一运行 task.run_forever()函数后 heap就溢出，，，，可能是什么原因造成的呢。。

这有可能和你所用的平台以及 rtt 的版本有关，请提供一些更详细的信息，关于你所使用的平台。

pikasTech, 2022-04-15:
已解决，原因：
rtt 的 shell stack 设置过小。导致 stack overflow

closed 2022-04-15 by pikasTech