pikasTech/PikaPython #139: Bluepill template built with arm-gcc does not fit 128K Flash

Chandler-Kluser, 2022-05-29:
I have got the [bluepill template for pikascript](http://pikascript.com/) and then made a Makefile to compile with arm-none-eabi-gcc, the code is in [this repository](https://github.com/Chandler-Kluser/pikascript_gcc_bluepill).

I was able to compile all .c files in the template into objects and them link them successfully, I could also build a binary firmware.

The issue is that the size is WAY too big for 128K bluepill flash memory, so I would like to know what objects may not be compiled to build a smaller binary.

Thanks in advance!

pikasTech, 2022-05-30:
maybe you can add the -Os option for size optimize.

https://github.com/pikasTech/pikascript/blob/562550dce02351b58f072f248fc79f23e7b331b7/bsp/stm32f103c8/Makefile#L25

Chandler-Kluser, 2022-05-30:
I believe it has fitted, but not sure because my STM32 is 64K:

```
arm-none-eabi-size bin/firmware.elf
   text    data     bss     dec     hex filename
 122660     492   19440  142592   22d00 bin/firmware.elf
```

pikasTech, 2022-05-30:
I'm trying to use the STM32CUBEMX auto-generated makefile to fix that.

For more details:

https://github.com/Chandler-Kluser/pikascript_gcc_bluepill/pull/1

Chandler-Kluser, 2022-05-30:
Compiled with the merge request you made, I have got:

```
arm-none-eabi-size obj/stm32f103c8.elf
   text    data     bss     dec     hex filename
  57264     484   18980   76728   12bb8 obj/stm32f103c8.elf
```

It would be great if there was ANYTHING that could be not used to make it fit in 64kB, but it worked in my PC!

I will also improve the Makefile you did, thanks a lot!

Chandler-Kluser, 2022-05-30:
I remember a week ago if we commented the line:

```python
import STM32F103
```
in main.py and recompile with pikascript compiler and then build the entire image again we could make the image fit in 64kB, but I am not finding that anymore in this new template.

closed 2022-05-30 by Chandler-Kluser

pikasTech, 2022-05-30:
> I remember a week ago if we commented the line:
> 
> ```python
> import STM32F103
> ```
> 
> in main.py and recompile with pikascript compiler and then build the entire image again we could make the image fit in 64kB, but I am not finding that anymore in this new template.

the STM32F103 is imported by mchine.pyi now.

Chandler-Kluser, 2022-05-30:
I did comment mchine.pyi file, recompiled that with pikascript compilar and then build the project again.

But the image did not get smaller, do you have any idea?

pikasTech, 2022-05-30:
> I did comment mchine.pyi file, recompiled that with pikascript compilar and then build the project again.
> 
> But the image did not get smaller, do you have any idea?

the 57k image is after commend the machin.py file.

if not commend the machin.pyi, the imagesize would be 71K

```
root@32d1708ead1f:~/pikascript_gcc_bluepill# ll bin/stm32f103c8.bin -h
-rwxr-xr-x 1 root root 71K May 30 11:30 bin/stm32f103c8.bin*
```

Chandler-Kluser, 2022-05-30:
Great, thanks a lot!!