pikasTech/PikaPython #274: ESP32 GPIO

laldana73, 2022-11-10:
Thanks for your awesome pikascript.
I am Testing with an esp32c3.
pikascript is running...
i have the pikascript shell, but i don't find how to turn on a GPIO pin.
is this functionality already implemented?
i am trying ...
io1 = ESP32.GPIO()
io1.setPin('5') here i have to put a string
io1.setMode('out)

here trow an error
[error] no match gpio port.

is GPIO implemented for ESP32? or do i have to implement the functionality?

Thanks Again.  Great Project

laldana73, 2022-11-10:
I foud the answer. the io1.setPin instruction is waiting 'Px' where x is the pin that i want to set.

pikasTech, 2022-11-11:
Further adaptations of esp32 are still in progress, including the network layer and Bluetooth. If there are no other problems, I will close this issue, and have a nice day!

closed 2022-11-11 by pikasTech