pikasTech/PikaPython #284: pika移植到esp32s3的lvgl的开发板里执行出错lv_checkbox1.py后经常会导致重启

net2uizoo, 2023-01-19:
pika移植到esp32s3的lvgl的开发板里执行出错lv_checkbox1.py后经常会导致重启
：

static void pika_task(void *arg)
{

 vTaskDelay(pdMS_TO_TICKS(1000));

pikaScriptShell(pikaScriptInit());
 

}

void app_main(void)
{
 printf("ruan:hello!\r\n");

//pikaScriptInit();

 
TaskHandle_t xHandle = NULL;
 
/*
 xTaskCreate(
                    pika_task,       
                    "pika_task",          
                    1024*20,     
                    ( void * ) 1,    
                    tskIDLE_PRIORITY, 
                    &xHandle );       

 xTaskCreate(
                    gui_task,       
                    "gui_task",          
                    1024*8,     
                    ( void * ) 1,    
                    tskIDLE_PRIORITY+2, 
                    &xHandle );   

*/

  /* 启动调度器 */
	//vTaskStartScheduler();
    // 如果程序运行到了这里就表示出错了, 一般是内存不足 

   xTaskCreatePinnedToCore(pika_task, "pika_task", 1024 *44, NULL, 1, NULL, 1);

   xTaskCreatePinnedToCore(gui_task, "gui task", 1024 * 44, NULL, 1, NULL, 0);
}

pikasTech, 2023-01-19:
请检查是否已经包含以下信息：
① 使用的平台、固件、版本
② 做过哪些尝试，什么结果
③ 完整错误日志

net2uizoo, 2023-01-19:
#
=============== [Code] ===============
import pika_lvgl as lv
import PikaStdLib
mem = PikaStdLib.MemChecker()

cb = lv.checkbox(lv.scr_act())
cb.set_text("Apple")
cb.align(lv.ALIGN.TOP_LEFT, 0 ,0)

cb = lv.checkbox(lv.scr_act())
cb.set_text("Banana")
cb.add_state(lv.STATE.CHECKED)
cb.align(lv.ALIGN.TOP_LEFT, 0 ,30)

cb = lv.checkbox(lv.scr_act())
cb.set_text("Lemon")
cb.add_state(lv.STATE.DISABLED)
cb.align(lv.ALIGN.TOP_LEFT, 0 ,60)

cb = lv.checkbox(lv.scr_act())
cb.add_state(lv.STATE.CHECKED | lv.STATE.DISABLED)
cb.set_text("Melon")
cb.align(lv.ALIGN.TOP_LEFT, 0 ,90)

print('mem used max: %0.2f kB' % (mem.getMax()))
print('mem used now: %0.2f kB' % (mem.getNow()))

=============== [File] ===============
[   Info] File buff used: 637/10240 (6.22%)
=============== [ Run] ===============
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200ad00  PS      : 0x00060730  A0      : 0x8200ad99  A1      : 0x3fcaddb0  
A2      : 0x3fc9b504  A3      : 0x0000000a  A4      : 0x3fc9b5a4  A5      : 0x0000000c  
A6      : 0x00000020  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3fcadd90  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0xffffe001  A13     : 0x00000000  
A14     : 0xffffe001  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000012  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  

Backtrace:0x4200acfd:0x3fcaddb00x4200ad96:0x3fcade00 0x42025e30:0x3fcade30 0x4205836d:0x3fcade80 0x42010fe1:0x3fcadea0 0x420110bd:0x3fcadec0 0x42007e3d:0x3fcadf00 0x4200938d:0x3fcadf30 0x4200a66c:0x3fcadf50 0x42008bfc:0x3fcadf70 0x42008c69:0x3fcadfb0 0x42008c4c:0x3fcadfd0 0x42008cd0:0x3fcadff0 0x4200caa9:0x3fcae010 0x4200fe99:0x3fcae040 0x4200ff47:0x3fcae060 0x42005f7e:0x3fcae080 0x40381989:0x3fcae0e0

net2uizoo, 2023-01-19:
用的esp32平台，idf4.4  ,

net2uizoo, 2023-01-19:
上电先执行 lv_btn1.py 后再执行lv_checkbox1.py，就正常，不会自动重启，
如果上电先执行lv_checkbox1.py 就会自动重启，

net2uizoo, 2023-01-19:
lv_btn1.py的文件内容为：

#!pika
import pika_lvgl as lv
import PikaStdLib
mem = PikaStdLib.MemChecker()

# create a simple button
btn1 = lv.btn(lv.scr_act())

btn1.align(lv.ALIGN.CENTER,0,-40)
label=lv.label(btn1)
label.set_text("Button")

# create a toggle button
btn2 = lv.btn(lv.scr_act())

btn2.align(lv.ALIGN.CENTER,0,40)
btn2.add_flag(lv.obj.FLAG.CHECKABLE)
btn2.set_height(lv.SIZE.CONTENT)

label=lv.label(btn2)
label.set_text("Toggle")
label.center()

print('mem used max: %0.2f kB' % (mem.getMax()))
print('mem used now: %0.2f kB' % (mem.getNow()))

#!pika

pikasTech, 2023-01-19:
用idf monitor查看panic的位置

https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4/esp32/api-guides/tools/idf-monitor.html

net2uizoo, 2023-01-19:
idf monitor下一行行打入lv_checkbox1.py的内容 正常，但是pikaStudio下拖入lv_checkbox1.py文件就重启，而且串口冲突，无法idf monitor

pikasTech, 2023-01-19:
> idf monitor下一行行打入lv_checkbox1.py的内容 正常，但是pikaStudio下拖入lv_checkbox1.py文件就重启，而且串口冲突，无法idf monitor

lv_checkbox1.py 复制进 main.py 然后预编译，刷固件

net2uizoo, 2023-01-19:
ELF file SHA256: 1f72fb5bee157965

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403756fc
0x403756fc: esp_restart_noos at G:/esp32/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/soc/esp32s3/system_internal.c:62

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x428
load:0x403b6000,len:0xbcc
load:0x403ba000,len:0x28d4
entry 0x403b61d8
ruan:hello!
======[pikascript packages installed]======
pikascript-core==v1.12.0 (2022/12/29 18:05:17)
ESP32==v0.1.0
PikaStdDevice==v1.11.0
PikaStdLib==v1.12.0
===========================================

~~~/ POWERED BY \~~~
~  pikascript.com  ~
~~~~~~~~~~~~~~~~~~~~
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200a940  PS      : 0x00060730  A0      : 0x8200a9d9  A1      : 0x3fcad440
0x4200a940: get_trans_style at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_style.c:555 (discriminator 3)

A2      : 0x3fc9acb4  A3      : 0x0000000b  A4      : 0x3fc9b014  A5      : 0x0000004c
A6      : 0x00000020  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3fcad420
A10     : 0x00000001  A11     : 0x00000000  A12     : 0xffffe001  A13     : 0x00000000
A14     : 0xffffe001  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000052  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
.

Backtrace:0x4200a93d:0x3fcad4400x4200a9d6:0x3fcad490 0x42025a70:0x3fcad4c0 0x4205685d:0x3fcad510 0x42010c21:0x3fcad530 0x42010cfd:0x3fcad550 0x42007a7d:0x3fcad590 0x42008fcd:0x3fcad5c0 0x4200a2ac:0x3fcad5e0 0x4200883c:0x3fcad600 0x420088a9:0x3fcad640 0x4200888c:0x3fcad660 0x42008910:0x3fcad680 0x4200c6e9:0x3fcad6a0 0x4200fad9:0x3fcad6d0 0x4200fb87:0x3fcad6f0 0x42005bbe:0x3fcad710 0x4038150a:0x3fcad770
0x4200a93d: get_trans_style at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_style.c:555 (discriminator 3)

0x4200a9d6: lv_obj_get_local_style_prop at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_style.c:294

0x42025a70: draw_indic at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/widgets/lv_bar.c:463

0x4205685d: _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1543 (discriminator 6)

0x42010c21: lv_txt_get_next_word at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/misc/lv_txt.c:193

0x42010cfd: _lv_txt_get_next_line at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/misc/lv_txt.c:309

0x42007a7d: lv_obj_get_y at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_pos.c:528

0x42008fcd: lv_obj_align_to at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_pos.c:401

0x4200a2ac: lv_obj_get_scrollbar_area at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_scroll.c:569 (discriminator 12)

0x4200883c: lv_clamp_width at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_pos.c:977 (discriminator 6)

0x420088a9: lv_clamp_height at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_pos.c:984 (discriminator 1)

0x4200888c: lv_clamp_height at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_pos.c:983 (discriminator 6)

0x42008910: lv_obj_refr_size at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_obj_pos.c:95

0x4200c6e9: refr_area_part at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/core/lv_refr.c:673

0x4200fad9: lv_style_register_prop at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/misc/lv_style.c:187

0x4200fb87: lv_style_remove_prop at g:\esp32\s3_demo_1213\lvgl_v8_pika\build/../components/lvgl/src/misc/lv_style.c:244

0x42005bbe: adjust_boot_time at G:/esp32/Espressif/frameworks/esp-idf-v4.4/components/newlib/time.c:63

0x4038150a: vTaskPriorityDisinheritAfterTimeout at G:/esp32/Espressif/frameworks/esp-idf-v4.4/components/freertos/tasks.c:4935

pikasTech, 2023-01-19:
看看是不是给 lvgl 开的资源（线程栈、堆）不足

net2uizoo, 2023-01-19:
xTaskCreatePinnedToCore(pika_task, "pika_task", 1024 *44, NULL, 1, NULL, 1);

   xTaskCreatePinnedToCore(gui_task, "gui task", 1024 * 44, NULL, 1, NULL, 0);

开了44K 呢

pikasTech, 2023-01-19:
看一下 lv_obj_style.c:555 附近的代码