pikasTech/PikaPython #291: rust-msc-latest-linux 工具运行时crash

teddyxiong53, 2023-03-01:
# 测试平台
5.4.0-136-generic #153~18.04.1-Ubuntu SMP Wed Nov 30 15:47:57 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
非docker环境。手动安装的相关工具链。手动编译。
# 现象描述
重复执行rust-msc-latest-linux 命令，如果目录已经有pikascript-api目录，则改命令会crash掉。此时需要手动删掉pikascript-api目录再执行则正常。
# 代码版本
fc5e21140b47e1247206ae2f47ea4ddcb5b888d5

# 出错打印
```
./rust-msc-latest-linux 
(pikascript) packages installed:
    pikascript-core==latest
    PikaStdLib==latest

(pikascript) pika compiler:
  scaning main.py...
    binding PikaStdLib.pyi...
    binding PikaStdData.pyi...
    binding PikaStdTask.pyi...
    binding PikaDebug.pyi...
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 21, kind: IsADirectory, message: "Is a directory" }', src/compiler.rs:53:40
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 5
Aborted (core dumped)
```

pikasTech, 2023-03-02:
我看了一下源码，就是rust操作fs的删除行为失败了。

有可能是权限问题，也可能是文件被占用了。试试加权限，或者用 wine 直接运行 exe

teddyxiong53, 2023-03-02:
今天又没有复现了。之前原因应该就是删除目录失败。可以加一个错误检查，出错时给提示并返回，避免crash。

closed 2023-03-02 by teddyxiong53