pikasTech/PikaPython #292: USB Communication Device Class (CDC) Serial/UART?

rog77, 2023-03-05:
Might you please consider adding in the functionality to allow USB CDC connection to REPL, maybe using the CherryUSB library? Now that Sipeed are pushing out more Bouffalo Lab  Bl702/BL616 boards that can run PikaPython via FreeRTOS*, this could increase your user base by making your software just a little easier to use out of the box.

 * https://github.com/bouffalolab/bl_mcu_sdk/tree/master/examples/pikapython

pikasTech, 2023-03-05:
Thank you for your feedback. You can override the following WEAK function to redirect REPL, including the USB CDC.

```
char pika_platform_getchar();
```

```
int pika_platform_putchar(char ch);
```

We are also working on porting Cherry USB to this project. Please stay tuned for our updates.

rog77, 2023-03-05:
Thanks for your response, I will keep an eye on this, it seems like a great project!

Is it a big job to do that over-riding? I am OK with python, not so much with C... I've run cdc examples from BL_SDK, but am not totally sure how it could like into your code.

pikasTech, 2023-03-05:
> Thanks for your response, I will keep an eye on this, it seems like a great project!
> 
> Is it a big job to do that over-riding? I am OK with python, not so much with C... I've run cdc examples from BL_SDK, but am not totally sure how it could link into your code.

The adapter of Bo Liu is what I am maintaining. I will prioritize using usb cdc to support repl, if the hardware supports it.

pikasTech, 2023-03-05:
the doc for bl616/618 http://pikascript.com/doc/board_sipeed_m0p.html you can get some info from it.

the english document will comming soon

pikasTech, 2023-03-17:
> Thank again for the response, I look forward to hearing the outcome of your efforts - much appreciated.

I'm currently merging support for USB_CDC into the mainline. You can find more information about this process in the following pull request: https://github.com/bouffalolab/bl_mcu_sdk/pull/120. Once the integration is complete, USB_CDC will become the default REPL port.

rog77, 2023-03-17:
That's great news, thank you!

If you're in the mood to do more USB stuff, might I suggest Host/USB-OTG HID for keyboard input?

I have a good reason for asking - the BL616 will be the onboard MCU programer for the Sipeed Tang Nano 20K FPGA. 

With this modification, and what you have already done I can see PikaPython being nearly a vaible firmware for the TN20K - this would open the door to all sorts of possibilities, not least PikaPython being adopted as an OS for a 'stand alone' computer with USB/HDMI...

...The TN20K will probably be quite popular as a retro-computer emulator, especially if PikaPython can control the FPGA - the BL616 chip is capable, it is all about the firmware - a device that can interactively select & flash bitstreams from onboard SD Card using MCU into FPGA SRAM via JTAG would be a game changer.

This, I am sure, would increase your userbase and add value for everyone.

pikasTech, 2023-03-18:
> That's great news, thank you!
>
> If you're in the mood to do more USB stuff, might I suggest Host/USB-OTG HID for keyboard input?
>
> I have a good reason for asking - the BL616 will be the onboard MCU [...]

Thank you for your suggestion and for sharing your exciting idea with me. However, I'm afraid I have to decline your request as my time and resources are limited, and I am not able to work on the Host/USB-OTG HID for keyboard input at the moment. Additionally, the Nano 20K's 616 firmware is not open for development.

While I understand the potential benefits of PikaPython being adopted as an OS for a 'stand alone' computer with USB/HDMI, I am unable to commit to any further development at this time. I appreciate your interest and support, and I wish you the best of luck with your project.

rog77, 2023-03-18:
Perfectly understandable. Thank you again for what you have already done! If I work out how to get HID going I will let you know. All the best.

closed 2023-03-18 by rog77