pikasTech/PikaPython #297: How to include python files as library in the build?

rog77, 2023-03-22:
https://github.com/micropython-Chinese-Community/mpy-lib/tree/master/lcd/I2C_LCD1602

^ as an example, I am looking to modify this code to run with PikaPython, I understand that there would need to be code changes to make it work - I selected this file because it is python only, and PikaPython supports I2C, so I won't have to do any C coding. 

But I can't see from the PikaPython Documents how to include this python file into the build process such that it can be imported as a library from another script or the REPL. I am using BL_MCU_SDK for Sipeed BL616DK.

Any advice/links/pointers are most welcome, thank you :-)

pikasTech, 2023-03-23:
[This document](https://pikadoc.readthedocs.io/en/latest/module_import.html) shows how to import a python module when compiling the firmware. Simply put, you need to run the pre-compiler: [rust-msc-latest.exe](https://github.com/bouffalolab/bl_mcu_sdk/blob/master/examples/pikapython/pikapython/rust-msc-latest-win10.exe). If you are using a Linux environment, you can use Wine to call this exe.

Just a side note, the support for i2c on BL618 is not yet fully developed. However, this issue will be improved in about a month because there will be an embedded competition using these features and more tutorials will be available then.

rog77, 2023-04-01:
Thanks for taking the time to get back to me. 

Thank you for the info, I will wait until it is fully implemented, I was hoping to modify that LCD driver for pikapython myself, but if the underlying protocol isn't fully working I would be less than useful.

Can I please ask, I am hoping to use SPI, SD card, and OS(directory: ls/cd file:open/read/write/close and the like) type functions using BL616, how far along is PikaPython in this regard?

I appreciate your help and advice.

pikasTech, 2023-04-10:
> Thanks for taking the time to get back to me.
>
> Thank you for the info, I will wait until it is fully implemented, I was hoping to modify that LCD driver for pikapython myself, but if the underlying [...]

Regarding your inquiry about the progress of PikaPython in terms of SPI, SD card, and OS-related functionalities, I'd like to direct you to our latest updates and discussions on this topic at https://verimake.com/d/285-bl618-pikapython-led-series-tutorial. This will provide you with the most up-to-date information on the status of these features.

Please feel free to join the conversation, ask questions, and share your insights. Your input and expertise are always welcome in our community. We look forward to your contributions and are excited to see PikaPython grow with your help.

rog77, 2023-04-18:
Thabks for your response :-)

closed 2023-04-18 by rog77