pikasTech/PikaPython #322: dict.items() set/list/tuple value segmentation fault & reverse items loop

n64rankaisija, 2023-06-25:
I'm not entirely sure on Python 3 version differences so I'm not entirely sure if these are considered issues for PikaPython. But I'm creating this issue just in case if these are considered issues.

I tested this code and it results in segmentation fault. It does not matter if it's using `()`, `{}` or `[]`, all of them seg fault.

```py
dict = {
    'a': {1, 2, 3},
    'b': {4, 5, 6},
    'c': {7, 8, 9},
}

for k, (a, b, c) in dict.items():
    print(k, ':', a, b, c)
```

Another thing that was unexpected to me was this going through the dictionary in reverse.

```py
dict = {
    'a': 1,
    'b': 4,
    'c': 7,
}

for k, a in dict.items():
    print(k, ':', a)
```

Output:
```
c : 7
b : 4
a : 1
```

pikasTech, 2023-06-25:
Thank you for bringing these issues to our attention.

I regret to inform you that the current version of PikaPython does not support some of the features you've tried to use:

The advanced tuple unpacking syntax  is currently not implemented. This might lead to unexpected behavior if employed in your code.for x, (x,x,x) in ...

Sets, as indicated by  syntax, are not supported at this time, meaning that any code utilizing sets may not execute as anticipated.{1, 2, 3}

Additionally, the order-preserving characteristic of dictionaries, a feature introduced in Python 3.7, is not yet supported in our platform. Consequently, when you iterate over a dictionary, the items may not be returned in the order they were inserted.

We recognize the importance of these features and our development team is actively working to incorporate them into future versions of PikaPython. For the time being, we suggest that you maintain this issue open. Once these features are integrated, I will personally update you in this thread.

In the interim, I recommend simplifying your code. For instance, when iterating over dictionary items, you could use a single variable like so: . Moreover, I suggest avoiding the use of sets and dependence on dictionary order to prevent any unpredictable behavior.for x, y in ...: a,b,c=y

Thank you for your understanding and patience. We truly value your feedback as it helps us improve our platform and better serve our users. Please don't hesitate to reach out if you have any further concerns or queries.

pikasTech, 2023-09-17:
fixed the reverse of dictionary iter.
[image]

https://github.com/pikasTech/PikaPython/commit/ca2820efb627b2bcb09eeddadb3e3b8e091e7fb0