pikasTech/PikaPython #323: invalid syntax for having comments inside

n64rankaisija, 2023-06-27:
Ran into this issue while I was messing around. Not a huge issue at all but thought that might as well report it just in case.

Having comments inside `info` and `dict` will result in invalid syntax.

This does not work:
```py
info = [
#   [ Age,   'Name',   'Place' ]
    [ 26,    'Jack',   'River' ],
    [ 32,    'Riven',  'England' ],
]

dict = {
#   'Name':  'Job'
    'Jack':  'Unemployed',
    'Riven': 'League of Legends'
}
```

This does work:
```py
#   [ Age,   'Name',   'Place' ]
info = [
    [ 26,    'Jack',   'River' ],
    [ 32,    'Riven',  'England' ],
]

#   'Name':  'Job'
dict = {
    'Jack':  'Unemployed',
    'Riven': 'League of Legends'
}
```

pikasTech, 2023-06-28:
to be fixed soon

pikasTech, 2023-06-29:
fixed https://github.com/pikasTech/PikaPython/commit/5da6b727e90080de0ab370a5375a073019d7ec54

closed 2023-07-01 by n64rankaisija