pikasTech/PikaPython #326: multiplication parsing bug

zwarrior1, 2023-09-15:
Hi I was bench marking  QUICKJS vs PIKAPYTHON (both are ported to CODEBLOCKS)
I encountered a weird case where following code causes segmentation fault in compiling stage:
```
a = 3.0
b = 1e-05 * a
```
but these ones do not cause errors
```
a = 3.0
b = 1e05 * a
```
```
a = 3.0
b = a * 1e-05
```
I'm not sure it is my fault (I misconfigured something) or Pika's internal bug.
please someone help me.

here is call stack dump if you need it:
```
#0 0x7ffff7d39e6e	_int_malloc(av=av@entry=0x7ffff7eafc80 <main_arena>, bytes=bytes@entry=24) (./malloc/malloc.c:3903)
#1 0x7ffff7d3b2e2	__GI___libc_malloc(bytes=24) (./malloc/malloc.c:3321)
#2 0x5555555b130e	pika_platform_malloc(size=24) (PikaPython/src/PikaPlatform.c:207)
#3 0x5555555b139d	pika_user_malloc(size=24) (PikaPython/src/PikaPlatform.c:223)
#4 0x5555555957b1	pikaMalloc(size=24) (PikaPython/src/dataMemory.c:51)
#5 0x5555555a14b0	New_PikaObj() (PikaPython/src/PikaObj.c:2477)
#6 0x5555555bcd09	New_TinyObj(args=0x0) (PikaPython/src/TinyObj.c:36)
#7 0x55555559d215	obj_newObjFromConstructor(context=0x555556208810, name=0x5555555c14c0 "", constructor=0x5555555bccf4 <New_TinyObj>) (PikaPython/src/PikaObj.c:463)
#8 0x55555559dac2	obj_getClassObj(obj=0x555556208810) (PikaPython/src/PikaObj.c:649)
#9 0x55555559d75a	obj_getProp(self=0x555556208810) (PikaPython/src/PikaObj.c:541)
#10 0x55555559d7b1	_obj_getPropArg(obj=0x555556208810, name=0x5555555c19d4 "__init__") (PikaPython/src/PikaObj.c:552)
#11 0x55555559d956	_obj_getMethodArg(obj=0x555556208810, methodName=0x5555555c19d4 "__init__", arg_reg=0x0) (PikaPython/src/PikaObj.c:610)
#12 0x55555559d9a4	_obj_getMethodArgWithFullPath(obj=0x555556208810, methodPath=0x5555555c19d4 "__init__", arg_reg=0x0) (PikaPython/src/PikaObj.c:619)
#13 0x55555559d9d2	obj_getMethodArgWithFullPath(obj=0x555556208810, methodPath=0x5555555c19d4 "__init__") (PikaPython/src/PikaObj.c:623)
#14 0x55555559eb09	obj_getNativeMethod(self=0x555556208810, method_name=0x5555555c19d4 "__init__") (PikaPython/src/PikaObj.c:1068)
#15 0x55555559f348	obj_runNativeMethod(self=0x555556208810, method_name=0x5555555c19d4 "__init__", args=0x0) (PikaPython/src/PikaObj.c:1293)
#16 0x55555559e0b9	_obj_initMetaObj(obj=0x555556208770, name=0x7fffff7ff450 "3294967297") (PikaPython/src/PikaObj.c:804)
#17 0x55555559e2d4	_obj_getObjDirect(self=0x555556208770, name=0x7fffff7ff450 "3294967297", pIsTemp=0x7fffff7ff4e0) (PikaPython/src/PikaObj.c:863)
#18 0x55555559e494	_obj_getObjWithKeepDeepth(self=0x555556208770, objPath=0x7fffff7ff52d "3294967297", pIsTemp=0x7fffff7ff4e0, keepDeepth=0) (PikaPython/src/PikaObj.c:904)
#19 0x55555559e51b	obj_getObj(self=0x555556208770, objPath=0x7fffff7ff52d "3294967297") (PikaPython/src/PikaObj.c:920)
#20 0x555555596132	queueObj_getCurrentObj(self=0x555556208770) (PikaPython/src/dataQueueObj.c:55)
#21 0x5555555a97c6	AST_parseSubStmt(ast=0x555556208770, sNodeContent=0x5555562086c8 "") (PikaPython/src/PikaParser.c:1490)
#22 0x5555555aaaea	AST_parseStmt(ast=0x555556208770, sStmt=0x555556208178 "1e-05*a") (PikaPython/src/PikaParser.c:1831)
#23 0x5555555a97d8	AST_parseSubStmt(ast=0x555556208510, sNodeContent=0x555556208178 "1e-05*a") (PikaPython/src/PikaParser.c:1490)
#24 0x5555555aab03	AST_parseStmt(ast=0x555556208510, sStmt=0x555556207f18 "1e-05*a") (PikaPython/src/PikaParser.c:1832)
#25 0x5555555a97d8	AST_parseSubStmt(ast=0x5555562082b0, sNodeContent=0x555556207f18 "1e-05*a") (PikaPython/src/PikaParser.c:1490)
#26 0x5555555aab03	AST_parseStmt(ast=0x5555562082b0, sStmt=0x555556207cb8 "1e-05*a") (PikaPython/src/PikaParser.c:1832)
#27 0x5555555a97d8	AST_parseSubStmt(ast=0x555556208050, sNodeContent=0x555556207cb8 "1e-05*a") (PikaPython/src/PikaParser.c:1490)
#28 0x5555555aab03	AST_parseStmt(ast=0x555556208050, sStmt=0x555556207a58 "1e-05*a") (PikaPython/src/PikaParser.c:1832)
#29 0x5555555a97d8	AST_parseSubStmt(ast=0x555556207df0, sNodeContent=0x555556207a58 "1e-05*a") (PikaPython/src/PikaParser.c:1490)
```

pikasTech, 2023-09-16:
This might be a syntax parsing error, I will check it.

pikasTech, 2023-09-16:
[image]
fixed https://github.com/pikasTech/PikaPython/commit/16242270c5aecb39be50d50aee472c336a27bc71

closed 2023-09-17 by zwarrior1