pikasTech/PikaPython #336: support Macos

pioneerAlone, 2024-03-14:
Hi ,

I am trying this library with Macos, but It seems that pikaPackageManager is only available in Windows. Is there a version or script that supports Macos/linux? In addition, is the source code open source?

pikasTech, 2024-05-02:
Here’s a refined response that specifies the untested nature of using PlayOnMac for this purpose on MacOS:

Hi,

Thank you for reaching out about pikaPackageManager on MacOS. Currently, pikaPackageManager is only supported on Windows. For Linux users, running Windows applications is possible using Wine. As for MacOS, one possible approach is to use PlayOnMac, a tool that relies on Wine to enable running Windows applications. However, please note that using PlayOnMac with pikaPackageManager on MacOS has not been officially tested and should be considered experimental.

Regarding the source code, it is not currently open-sourced, but we are open to the possibility of making it available in the future. We acknowledge your suggestion regarding the development of a MacOS version using Rust's cross-platform capabilities, especially for the M2 (ARM64) architecture. We appreciate your feedback and will take it into consideration as we plan future updates.

pioneerAlone, 2024-05-23:
Thanks a lot, will follow this.

closed 2024-05-23 by pioneerAlone