pikasTech/PikaPython #342: Can't start Docker

pioneerAlone, 2024-06-21:
I am trying to pull this library and test docker, but there is a issue when I execute "bash build.sh"
Any advise here?

`# bash build.sh
Error response from daemon: No such image: pikadev:latest
[+] Building 33.2s (2/2) FINISHED                                                                                                                        docker:default
 => [internal] load build definition from Dockerfile                                                                                                               0.1s
 => => transferring dockerfile: 2.71kB                                                                                                                             0.0s
 => ERROR [internal] load metadata for docker.io/library/ubuntu:22.04                                                                                             32.5s
------
 > [internal] load metadata for docker.io/library/ubuntu:22.04:
------
Dockerfile:1
--------------------
   1 | >>> FROM ubuntu:22.04
   2 |
   3 |     ADD ubuntu22_sources.list /etc/apt/sources.list
--------------------
ERROR: failed to solve: ubuntu:22.04: failed to resolve source metadata for docker.io/library/ubuntu:22.04: failed to copy: httpReadSeeker: failed open: failed to do request: Get "https://production.cloudflare.docker.com/registry-v2/docker/registry/v2/blobs/sha256/67/67c845845b7de8024a1ad9f6e7fd08964502a0b423aa8de631ef521863873884/data?verify=1718952756-5B7zLTqu3X9DtFQbHshjcrVMM10%3D": dial tcp 108.160.163.106:443: i/o timeout
root@q-OptiPlex-5050:/home/q/bakewell/pika-python/pikapython/docker#
`

pikasTech, 2024-06-21:
网络问题

pioneerAlone, 2024-06-24:
好的，我再检查下

closed 2024-06-24 by pioneerAlone