pikasTech/PikaPython #344: pikaVM_runSingleFile with a error

pioneerAlone, 2024-07-02:
Hi ,

I  encountered a problem in executing pikaVM_runSingleFile(pikaMain, "test2.py") on windows with mingw ENV.
The test2.py file is located in the pikapython directory，

 ```
   // run single file
    PikaObj* pikaMain = pikaScriptInit(); // no dependancy, need precompile
    pikaVM_runSingleFile(pikaMain, "test2.py");
```
Test to run single file with a error below:

**here in main.py
Assertion "NULL != file_arg" failed, in function: pikaVM_runSingleFile().
  (at pikapython\pikascript-core\PikaVM.c:4024)
error: execv(E:\Pikapython_demo\build\mingw\x86_64\release\pikapython.exe ) failed(3)**

pikasTech, 2024-07-03:
The root cause of this issue is that the test2.py file needs to be located in the same directory as the pikapython.exe executable, not in the pikapython directory.

Please ensure that test2.py is placed in the execution path of pikapython.exe. This should resolve the issue of the file not being found.

pioneerAlone, 2024-07-03:
Thanks, 

It's working now.

closed 2024-07-03 by pioneerAlone