pikasTech/PikaPython #350: [技术咨询] 是否支持脚本里在嵌套加载新脚本以实现热更新的功能

24gdolphi, 2024-11-01:
### 前景
目前在支持做一个杰理的项目，客户想解决应用无法更新的痛点，看到资料了解到PikaPython，这是很nice的项目，想着是否能支持到类似热更新的功能

### 咨询
之前了解过python在python-pc端，我可以实现类似下面这样的，其中 code_str 的内容我可以通过网络下载获取到，我的 main.py只负责check和download脚本，想问下在PikaPython里有实现类似的方法吗，如果有的话大概是怎么样实现的，感谢🤝

[image]

pikasTech, 2024-11-02:
PikaPython直接支持exec，这个没问题，网络和文件系统的访问也是支持的（需要对硬件做适配）。
运行速度不如lua，lua的虚拟机更简单，具体速度和平台有关。

24gdolphi, 2024-11-04:
谢谢回复，我拉取了代码暂没找到关于类似exec的api函数或者例子，能否告知下相关的位置😊

pikasTech, 2024-11-07:
exec 是pikapython提供的内置函数，可以直接使用

https://github.com/pikasTech/PikaPython/blob/541b9a974351526d1e07aa3bf6bd5ee93fc90b1a/package/PikaStdLib/builtins.pyi#L73

24gdolphi, 2024-11-07:
谢谢支持🤝👍

closed 2024-11-07 by 24gdolphi